bluerobotics/product-documentation #25: BROV2: Plug in/turn on rov should come BEFORE network setup

Williangalvani, 2019-04-30:


jaxxzer, 2019-06-26:
When network setup comes first (like it does today), users may configure the laptop's built-in network interface instead of the transient usb-ethernet interface.

danielgud, 2020-02-07:
The step for plugging in the ROV/FXTI is in the "Topside Setup" section of the Assembly guide, which then refers to the FXTI documentation. 

Should topside setup be moved to the beginning of the Software setup guide?

danielgud, 2020-03-20:
@bluerobotics/software-team a bump for my previous question:

> The step for plugging in the ROV/FXTI is in the "Topside Setup" section of the Assembly guide, which then refers to the FXTI documentation. Should topside setup be moved to the beginning of the Software setup guide?

patrickelectric, 2020-03-20:
> Should topside setup be moved to the beginning of the Software setup guide?

I believe that should stay in the assembly page, there is a reference in [the software setup page](https://bluerobotics.com/learn/bluerov2-software-setup/) for the assembly guide, where it ends with FTDI/Fathom-x connection.

Maybe after:
> These are instructions for the first time that you set up your computer to work with the BlueROV2. If you have not assembled your BlueROV2, please see our Assembly Guide and assemble your BlueROV2 prior to setting up the software.

We should add:
> Make sure to sure to have your ROV powered and with the FTDI/Fathom-x before proceeding with the instructions. 

@Williangalvani  @jaxxzer any input about this ?

danielgud, 2020-03-25:
Added this note to the software setup guide:
[image]

closed 2020-03-25 by danielgud