bluerobotics/product-documentation #31: Add i2c communication doc

jaxxzer, 2019-04-30:
Things about wire distance, signal integrity, speed, logic levels, etc.

kklemens, 2020-03-03:
Thinking like a "Basics" guide or somewhere else?

rjehangir, 2020-04-01:
Are we getting many questions about this? We could probably just make sure the important stuff (wire length, logic levels) is mentioned on the relevant product pages and link to a Sparkfun or Wikipedia article that provides more details about the I2C protocol.

kklemens, 2020-04-01:
We're not getting any questions about this. Ok, we can write one and then stick it on every product page that uses I2C?

rjehangir, 2020-04-01:
Okay. Or just make sure this is already clear on those products. I'm fine closing this issue as long as we double check that.