bluerobotics/product-documentation #35: SEO focus

jaxxzer, 2019-05-09:
Attempt some basic SEO.

For example when google searching most of our iconic products ie 'fathom s' or 'bar 100', the product page should be the first hit. Instead, some of the product pages do not appear at all in the results, and in many cases the documentation, tutorials, and forum posts come up first!

Williangalvani, 2019-05-09:
A great tool for that is the Google Search Console. I believe @rjehangir is probably the owner of the website there.

jaxxzer, 2019-05-14:
'blue robotics desiccant' results in the retired product

Williangalvani, 2019-05-14:
> 'blue robotics desiccant' results in the retired product

And the retired product doesn't point to the new one.

jaxxzer, 2019-05-14:
same story with the lumen light..